fesch/CanZE #757: CanZe on laptop/PC

Ely7777, 2022-09-27:
Hello,
First of all, thank you very mug fir creating this app. 
I’ve run into a series of issues when trying to use CanZe in my laptop (Windows, not Mac). 
I’ve tried several Android emulators like Bluestack, Memo, etc, but there is no way that I can connect CanZe with the OBD2 (ELM327) via Bluetooth. I click the icon and it just won’t work. 

I’d really appreciate some feedback on thus issue, thanks!

KarelSvo, 2022-09-27:
Pyren is better suited for Renault fault diagnosis.  https://gitlab.com/py_ren/pyren/-/blob/master/README.md

Ely7777, 2022-09-29:
Hello, thank you for your answer. Are you saying that there is no other way that I can use CanZe on my laptop? As I've been looking at Pyren but it's all Greek to me. Thanks

yoh-there, 2023-10-02:
Seems like Win11 can do some android emulation, as can a chromebook. However, I did not get BT communication to work on my Chromebook. Closing for "won't fix"

closed 2023-10-02 by yoh-there