fesch/CanZE #758: Connection timeout R110 LELink2 OBD-II

Hvglaser, 2022-09-30:
I have a Bluetooth OBD connector in the port of my R 110.
My phone recognizes the connector(it's visible and connected in the Bluetooth settings on my iPhone), and it worked on my old Zoe, a Q210.

But I'm getting connection time out when I'm trying to connect using the latest TestFlight version of the app. I've tried all the different settings, but I am stuck. any ideas?

ik2wxx, 2022-09-30:
Hello
I see no reason that app does not connect to new auto, if the adapter is compatible with the app, connected correctly to car and app is configured correctly.
Does any other similar app work with same car and adapter?
Roberto

ik2wxx, 2023-10-10:
Closed for no activity

closed 2023-10-10 by ik2wxx