fesch/CanZE #761: Tires astp6 error.  Cannot write new values

newbyr, 2022-11-29:
Hello 
Car is Zoe R90
Canze v1.60
Adapter Konwei 

I know this is a long runner, but I've read the previous threads and csn see no help for this now.
I have changed from summer to winter tires.  In  the summer I recorded a screenshot of the codes with Canze, and then paid Renault to change them.  I also loaded then to save set A.  But neither recalling the saved data, or manually entering the codes works.
Now changing back, I have tried to reload the data with no success. I have tried with and without ISO on in settings.  No effect.

Screenshot is attached.
I get an error for each tire of :
E-problem sending atsp6 command 

Any help is appreciated.
Richard 

[image]

[image]
[image]

yoh-there, 2022-11-29:
Thank you for the details. But this is a very strange error, as atsp6 is a dongle initialization command and things have not reached a state of actually trying to write, so even if ie the car had no TPMS at all, it should still not fail on this command and if I am not mistaken it is also send on CanZE initialisation. No other issues with it?

Unfortunately, I do not have a car with TPMS myself so I cannot test it (wrote the code years ago and using somenbody elses car). I hope others can confirm it works, which I think it does; it has been in here for ages.

Any chance of trying another dongle? Even these ones are unfortunately cloned :-(

Apologies for the vague answer.

newbyr, 2022-11-29:
Hi, Thanks for the reqply.

Unfortunately, I only have a basic (Ebay) OBDII other than the Konwei, but .....

I tried the basic and got the same messages.  It managed to check through the rest of the car and found everything else, and passed the CanZe ELM test (both with and without ISOTP fields).  

I'm starting to wonder if the cunning engineers at Renaulthave put a block on the data so that I have to spend my money with them to get the codes swapped ??

closed 2023-10-02 by yoh-there