fesch/CanZE #764: Question about power regenaration

Snooone, 2023-01-19:
Hi,
I am pretty new to this forum on CanZE, and I don't know where to place this question, so I post it on github Issues wall. please let me know where to post it, if it is not i the right place.

For some student study, I would like to know if CanZE is able to export the following raw data into an android text file : 
- Time-Stamp
- Instant Speed
- Instant regenerated Power or Consumed Power (or Instant Voltage / Intensity)
- ... or any other physical variables that are available from the Dongle DataBus.

Many thanks in advance for your help
Serge

NLFiscaal, 2023-01-19:
Hello Serge,

Have a peek here
https://canze.fisch.lu/screen-for-researchers/

yoh-there, 2023-01-19:
Hello Serge,

Have a peek here
https://canze.fisch.lu/screen-for-researchers/

Snooone, 2023-01-19:
Thx Yoh, very good!
it rises some further (beginner) questions:
- does the fields definition of ZOE Ph2  fits with a ZOE 2021 R135?
- are there timestamps against each measure ?
- and where to find the extended names/definition of the controllers like ABS-VDC which certainly stands for the ABS controller ?

yoh-there, 2023-01-20:
>     * does the fields definition of ZOE Ph2  fits with a ZOE 2021 R135?
AFAIK, yes
> 
>     * are there timestamps against each measure ?
Yes. You can also play around with the log settings in the settings screen, but "Research" is the most flexible.
> 
>     * and where to find the extended names/definition of the controllers like ABS-VDC which certainly stands for the ABS controller ?
What we have is in the assets files. If I remember correctly, the only one we didn't include is the Airbag computer, for obvious reasons.

If you want to do serious logging, I would advice to consider building the CanSee dongle as it is much faster. Link: https://gitlab.com/jeroenmeijer/cansee/

Snooone, 2023-02-10:
Many thanks @yoh-there, I have just received the OBD2 and am now on the way to use it with CanZE.

Snooone, 2023-06-02:
Hello @yoh-there, back again on the subject,
we succeeded to log values through the Research wizzard using a Standard OBD2Bluetooth.

We have 2 points we would like to improve:
- the logging rate is more than 1 to 2 s, which is not sufficient
- we are looking to log the instant power consumed or saved (thru regeneration breaking) at the battery side. Any idea which field could provide this information ?

thanks in advance

here the values we are currently logging on a ZOE Phase 2 50kwh
,762,24,31,1,0,0,km/h,220118,620118,ff,FilteredVehicleSpeed
,18daf1da,24,39,.25,2000,1,A,2221cc,6221cc,ff,High voltage battery Current measured consolidated
,18daf1da,24,39,.001,0,0,V,999999,999999,ff,Voltage measurement given by BCS Battery Current Sensor

yoh-there, 2023-06-02:
Hello @Snooone ,

As for speed, you really need to check out the CanSee project that I meantioned earlier. It is MUCH faster.

For power, our strategy is to obtain DC voltage and current. Sure this then includes non-traction consumers and losses, but that should be consistent enough (IMHO of course). I am fairly sure there is a virtual (computed field for that you can use directly.

Hope this helps.

closed 2023-10-02 by yoh-there