fesch/CanZE #765: Zoe Z.E. 50: TPMS (tires) section not working

pciori, 2023-01-24:
Hello.
I have a Z.E. 50 (R110) and a Konnwei KW902 connected with Bluetooth to an Android phone (Android 9). The car is equipped with TPMS allowing to read on the car dashboard all the four tire pressures.
The CanZe app is set to ZE40/ZE50, ISOTP OFF or ON (no change) and when I try to enter to the tires section of the CanZe app it is grey and no data are populated. Even waiting for a long time (driving) no data is populated.
In the bottom line of the screen the commands are displayed and apparently there is no issue.
All the other sections like Battery, Charging, Temperature heatmap, Voltage heatmap are working properly. Some ECU are not accessible and errors are shown or timeout displayed.
Am I doing anything wrong?
Next week I need to swap my front/rear tyres and I would like to use the option for doing that in CanZe.
Greetings 
Paolo

[image]

ik2wxx, 2023-01-24:
Ciao Paolo.
Sulla ZE50 la parte relativa ai sensori di pressione dei pneumatici è disattivata, perché tutte le funzioni vengono fatte dal display dell'auto, non c'è quindi bisogno di usare CanZE per questo.
Attendo comunque la risposta di @Notice-Jeroen-Meijer in merito.

yoh-there, 2023-01-24:
Totally correct answer (of course!) by @ik2wxx . No need to do anything with CanZE on a ZE50. Closing this issue, feel free to reopen if there are more questions.

closed 2023-01-24 by yoh-there

pciori, 2023-01-24:
Ciao Roberto,
grazie per la risposta. Sì posso leggere dal display dell'auto le pressioni, ma non gli ID dei sensori TPMS, né settare i nuovi ID quando devo cambiarli.

pciori, 2023-01-24:
@yoh-there 
You are right, but from the car dashboard I can just read and reset the tire pressures, not the TPMS sensors ID and write their new ID in the car computer when I will change them in the near future.

yoh-there, 2023-01-24:
You can "learn" the car new TPMS sensors in a ZE50, 100% sure. I am not sure _how_ as I do not own a ZE50; I think I heard it will actually relearn automatically after a couple of minutes driving around, but again I am not 100% sure of that.

pciori, 2023-01-24:
Unfortunately that is not possible as the receiver is only one and it is not possible to assign the correct front/rear/left/right ID to the car computer. The only option to do that is to ask to an authorized Renault dealer.

yoh-there, 2023-01-24:
That is correct. For the TPMS alarm light I guess they deemed it "good enough".