fesch/CanZE #768: CanZE charging history problem

MatteoTorba, 2023-03-08:
Hi. After a zoe lbc update, in CanZE app now i have all charging history parameters reset. Is all 0. Someone can tell me why?
Thank you

fesch, 2023-09-28:
Some other users reported that the latest version (1.67) works, expect for the charging graphs. It sounds strange, that some values a read and other are not, but if this is related to an update of the LBC, it might not be a CanZE problem and thus related to what others say .... BUT, version 1.63 is reported to still work. Could you try that?

yoh-there, 2023-10-02:
Graphs work again. History works on my Q210. If this is still an issue, please state car type and build year and we'll see what we can do. If not (in 1.68), please close.

MatteoTorba, 2023-10-02:
Hi! 1.68 works great. Thank you!

Inviato da Yahoo Mail su Android

closed 2023-10-03 by yoh-there