fesch/CanZE #770: Konnwei with Canze used to work but now doesn't

usernamepasswordBS, 2023-03-12:
Hi, I have a Konnwei BT dongle with the Canze app installed on my phone. A few months ago it worked fine but now it won't connect via bluetooth. I have tried another phone, another car and another app. The other OBD app will connect to the Konnwei and also work with the car in question. 

So I have found that it's not the phone, nor the dongle nor the car but it must be the app itself.

What can I do?

fesch, 2023-03-12:
Did you update the App?

usernamepasswordBS, 2023-03-12:
Hi, yeah I think so. I mean, I deleted it and re-installed it.

fesch, 2023-03-13:
That's strange ... so if it's not the car, nor the dongle, nor the phone and the app is the same, it must be something else. Maybe some external source that has some interference with the BT of the dongle?

usernamepasswordBS, 2023-03-13:
I tried it in my other car and in the Zoe parked in different places so I can't think what might be interfering. To give you more detail: a year ago I had a different phone and I installed Canze and it worked fine back then. I changed my phone around Christmas and didn't install Canze until last week when the car wouldn't charge. It didn't work so I installed Canze on my old phone and it was the same so I moved onto the next steps that I have already described. I am wondering if it's possible to install an older version to see if there is a conflict between the newer version and the dongle or even my phone(s).

fesch, 2023-03-14:
I have a very old version over here. Maybe you can test that one:

https://www.fisch.lu/junk/canze.apk

usernamepasswordBS, 2023-03-14:
> From "Bob Fisch" ***@***.***>
> To "fesch/CanZE" ***@***.***>
> Cc "usernamepasswordBS" ***@***.***>; "Author"
> Date 14/03/2023 14:54:03
> Subject Re: [fesch/CanZE] Konnwei with Canze used to work but now
> doesn't (Issue #770)

Hi, thanks very much for sending that unfortunately it keeps saying "App 
not installed".

fesch, 2023-03-24:
Hmmm ... that's strange. Then there must be something else that blocks the installation, as it should generally work out of the box.

usernamepasswordBS, 2023-03-24:
Hi Bob,

Thanks again for responding. I have tried on another phone and I get the 
same problem with "app not installed".

The up to date version installs but on connecting to the Konnwei via 
bluetooth, immediately disconnects. That also occurs on the second phone 
that I tried, one is a Samsung and the other a Wiko.

Date 24/03/2023 08:35:08

fesch, 2023-09-28:
What is the latest version you tried?

usernamepasswordBS, 2023-09-28:
Latest version of CanZe ?  I can't remember which but I've bought a Vgate iCar Pro 2.3 and it worked fine.

Thanks,

Date 28/09/2023 18:45:20

closed 2023-10-06 by yoh-there