fesch/CanZE #771: App incompatible with Moto G62

BooMeringue, 2023-03-20:
Hello,

I am trying to install CanZE on a Moto G62 running android 12 and Play store tells me it's not compatible with my device. The phone is completely vanilla, no weird software hacking or anything.

I've not seen this issue before with any other apps. Any ideas?

Many thanks,

Toby

fesch, 2023-03-20:
I was contacted by someone else with the same problem on another device which occurred after a full reset.

Could you try to install the APK manually?

https://www.fisch.lu/junk/app-playstore-release.apk

Notice-Jeroen-Meijer, 2023-03-28:
@fesch Maybe there is a larger problem here. I just got a message from a Dutch user unable to install on a Nokia 6.2 om Android 11.

yoh-there, 2023-03-28:
@fesch Maybe there is a larger problem here. I just got a message from a Dutch user unable to install on a Nokia 6.2 om Android 11.

fesch, 2023-03-28:
@Notice-Jeroen-Meijer : indeed :-(

BooMeringue, 2023-03-28:
Bob,

Sorry for the slow reply. I was able to install the app using the apk, but
it crashes every time I try to go into "settings" which obviously is needed
to link to the bluetooth dongle. I noticed there was a clone app called
'CanZE plus' on the play store which works fine.

Happy to continue testing if you need me to.

Toby

-- 
Toby Cornwall
Mechanical engineer
T: +44 (0)7984491898

fesch, 2023-03-28:
"CanZE plus" is a fork, nothing more ... we need to check what the problem is. I need some more recent device I think ;-)

yoh-there, 2023-10-02:
I think this has been resolved? If so, please close. If not, add a comment please. Will close for being stale in a short while.

BooMeringue, 2023-10-17:
For my part, I just switched to using CanZE+ so I can't comment on whether
the issue is fixed on CanZE.

Toby

-- 
Toby Cornwall
Mechanical engineer
T: +44 (0)7984491898