fesch/CanZE #773: iOS TestFlight version not showing tires data, but App Store version (3.0.1.0) does show it

tomgru, 2023-04-05:
Hello! I have a Zoe R240 and I am now using the iOS TestFlight App.

When I go to the tires menu, I see lots of debug info on the bottom of the page. It keeps showing data like "Debug 765.6171.16 ok 1.00" or "Debug 673.0 ok 0.00" or "Debug 673.16 EMPTY 2539.12", but nothing else happens on that page. E.g. the tire pressures are never shown even if I wait 10 minutes.

I have still the previous CanZe version from the Apple App store installed and there at least the tire IDs are shown almost instantly.

But I need the new App version, because I want to use it for re-programming the tires when changing them in the future by myself.

Many thanks in advance for looking into this issue. Let me know if there is anything I can try in addition.
tom

ik2wxx, 2023-04-05:
I'm sorry, that page has never been tested on a real car by anyone of the developer team.
Personally I have access only to a ZE50 and that screen is not needed in that car model.

closed 2023-04-05 by ik2wxx

tomgru, 2023-04-05:
interesting. on https://canze.fisch.lu there are two posts that say otherwise:

https://canze.fisch.lu/save-tire-set/
https://canze.fisch.lu/tires-tpms-revisited/

Could it be that this is only working on Android but not yet on iOS?

tomgru, 2023-04-05:
thanks & apologies asking, but do you know when that feature could become available on iOS?

ik2wxx, 2023-04-05:
I'm sorry I don't know anyone with and old Zoe :-(