fesch/CanZE #776: Initialization Error on VGate iCar Pro

ProgrammingHobby, 2023-05-07:
Can not Init with Vgate iCar Pro BLE4.0 ELM327 v2.3 Firmware 4.1.10

mgottschlag, 2023-05-07:
What a coincidence. See my other bug report a couple of minutes ago. Do you also see the "atws" error message?

ProgrammingHobby, 2023-05-07:
Initialisierung gescheitert
Problem: Unrecognized ELM version response [atws]

fesch, 2023-05-10:
Please test your dongle with the acid script and report if it works. Mostly the output of the version numbers are *faked*.

https://canze.fisch.lu/dongle-test-script/

sanchosk, 2023-06-21:
Not sure it helps to someone, but I found out this:
I have similar dongle, also shows as 2 devices (IOS-Vlink and Android-Vlink).
When I try to connect with CanZE, I get atws error.
Then, I start carscanner, connect to the same dongle, carscanner works OK, but reads only battery voltage, etc (old Zoe, it does not have correct protocol).
However, after I quit carscanner and start CanZE again, this time everything connects, I can read state of charge, charging speed, etc, no problem.
Is there some data I can provide to devs to investigate and potentially fix the issue? I guess there is some ELM327 reset or something that the carscanner does perform and CanZE does not.

sanchosk, 2023-06-21:
Here is the log from carscanner app:
`
[Recorder hasData=false]
===== LOG STARTED: 21.06.2023 12:03:31=====
=====Version=1.99.1 Build=400991 P=False NoDelayELM327Init=False
====Connection profile=Renault Zoe Phase 2
[Connection type: Bluetooth; DeviceName=Android-Vlink]

[Connecting to B4:88:C2:17:5B:94]
bt_socket.Connected
got streams from client

[Connected to B4:88:C2:17:5B:94]
ATZ
ATE0

[ELM RESET]

ELM327 v2.3

>
Initialize(initMode=Default)

InitializeCustomInitString(initMode=Default)
ATE0
ATE0
OK

>STI
?

>VTI
IcarPro v2.3

>
[ELM RESET]
ATZ

ELM327 v2.3

>ATE0
ATE0
OK

>ATH1
OK

>ATSP6
OK

>ATS0
OK

>ATM0
OK

>ATAT1
OK

>ATSH7E4
OK

>222002
7EC056220020FB0

>222002
7EC056220020FB0

>
[FUELRATE: FuelRate;FuelRate=123:False:Engine fuel rate;MAF=16:False:MAF air flow rate;MAP=11:False:Intake manifold absolute pressure;IAT=15:False:Intake air temperature;RPM=12:False:Engine RPM;AFR_LAMBDA=92:False:Fuel/Air commanded equivalence ratio;FSS=null;LOAD_ABS=91:False:Absolute load value;LOAD_PCT=4:False:Calculated engine load value;Economizer=229:False:Fuel economizer (based on fuel system status and throttle position);BARO=67:False:Barometric pressure;AMBIENT_TEMP=94:False:Ambient air temperature;O2_V1=null;O2_V2=null;INJECTOR=122:False:Fuel injection timing;CYCLE_CONSUMPTION=null;]
[... 3439 lines cut ...]
>222002
7EC056220020FEB

>222002
7EC056220020FEB

>222002
7EC056220020FEB

>222002
7EC056220020FEB

>222002
7EC056220020FEB

>
Stop(Disconnect:UserClickbtnDisconnect)

[SetRunning(False,From Stop #3640)]

Stopped

===== DISCONNECTED AT: 21.06.2023 12:07:41

`

yoh-there, 2023-10-02:
Closed (stale)

closed 2023-10-02 by yoh-there