fesch/CanZE #778: Timeout Read iOS Vgate

BraxNL, 2023-06-19:
When I try to connect to Canze using Vgate i-car pro I can sort off connect (app first says I’m connected) but after a few second the message “TIMEOUT read” appears, not data is being shown. The Vgate stays blinking blue with random one or two green flashes.

When I try to connect with with a different app, for example “Auto Doctor” the connection is perfect with a steady green light on the Vgate.

What could this be and how to fix it? Help would be very much appreciated!

ik2wxx, 2023-06-20:
Hello, I have 2 Vgate and they work well. Maybe something has changed in new models.
Please scroll the app main screen to the last page, you will see a "TEST OBD" button.
Use that screen to discover correct OBD connection parameters, then copy/paste the log text here.
Thank you

Sofille65, 2023-06-20:
Same error on iOS with LeLink via Bluetooth

KarelSvo, 2023-06-20:
> When I try to connect to Canze using Vgate i-car pro I can sort off connect (app first says I’m connected) but after a few second the message “TIMEOUT read” appears, not data is being shown. The [...]

https://github.com/fesch/CanZE/issues/775

ik2wxx, 2023-06-20:
> #775

That thread is about Android app, not iOS

KarelSvo, 2023-06-20:
> > 
> > #775
> 
> That thread is about Android app, not iOS

https://github.com/fesch/CanZE4iOS/blob/main/CanZE/CanZE/_devices/ELM327.swift#L86
For me the problem looks the same. ATWS timeout = 0 = let MINIMUM_TIMEOUT = 100 
icar pro needs a little more than 120 ms for the response of the elm version.

riker82-blip, 2023-06-22:
Same issue for me too (iOS with vGate iCar Pro) since the TestFlight invitation code has been renewed. Can you please help?

riker82-blip, 2023-06-22:
> Hello, I have 2 Vgate and they work well. Maybe something has changed in new models.
> Please scroll the app main screen to the last page, you will see a "TEST OBD" button.
> Use that screen to discover correct OBD connection parameters, then copy/paste the log text here.
> Thank you

Test OBD fails as well with the same Timeout Read error

spr97ajm, 2023-07-19:
I have this same issue.

Geekfried, 2023-08-14:
Same issue for me :-/

thymjan, 2023-09-11:
Same here. Connection to dongle is established. But data transfer is interrupted because of timeout with (Testflight) iOS app 1.0.1(18).

With android app data transfer is working flawlessly.

Even with ABRP (iOS) data transfer iis working.

[image]
[image]

jorismw, 2023-09-13:
Same here, such a shame :(

Agalash, 2023-09-17:
Hello all,

I just bought a VGate BLE and I have the same issue.
Do you know if this problem is specific to this dongle or it’s an IOS bug?

druelie, 2023-09-17:
In "A Better Route Planer" it works flawlessly, so it cannot be an iOS problem.

Sofille65, 2023-09-17:
> Hello all,
> 
> I just bought a VGate BLE and I have the same issue. Do you know if this problem is specific to this dongle or it’s an IOS bug?

It is not a dongle problem but the latest version of CANZE is not working properly under iOS. We asked for a roll back but it seems that iOS users are not a priority. So I am not using the app since months

ik2wxx, 2023-09-18:
> It is not a dongle problem but the latest version of CANZE is not working properly under iOS. We asked for a roll back but it seems that iOS users are not a priority. So I am not using the app since months

Hello all
It's not a priority problem, it's a "car availability" problem.
Will work on that when I will have the car for testing, sorry It's taking so long.
I hope to work on a solution before the end of this month.
Roberto

Sofille65, 2023-09-18:
> Hello all It's not a priority problem, it's a "car availability" problem. Will work on that when I will have the car for testing, sorry It's taking so long. I hope to work on a solution before the [...]

Why not putting the previous iOS version? 
this one was working fine

ik2wxx, 2023-09-25:
Hello all, new version is available. 
No changes from previous versions, just a restore from build 17. 
Please let us know.

druelie, 2023-09-25:
It works again!
Thank you!

Agalash, 2023-09-25:
Thanks for this version
It works :-)

ik2wxx, 2023-09-28:
If running into connection troubles, please scroll to the last page in app's main screen, then tap on "TEST OBD". 
It's a special OBD testing page, you can make connection and dialog experiment with your dongle, than if you want you can report results on this page.
Thank you

closed 2023-09-28 by ik2wxx