fesch/CanZE #779: Bluetooth 5.0 Konnwei kw902

henrikx, 2023-06-19:
Today I received a Konnwei kw902, but it seems to be a very new version with Bluetooth 5.0 (Low Energy LE) connectivity instead of Classic (3.0). Even if I am able to pair with it in Android settings, the app just doesn't want to connect with the dongle. Bluetooth icon in app is blue with connecting symbol and stuck like that. 

Does CanZE just not support connecting to dongle over Bluetooth LE? Other OBD apps are able to connect and communicate, however we all know why we're here using CanZE instead of those apps.

Any help appreciated.

zaelu, 2023-07-22:
same problem for me too. bought a BT 5.0 because couldn't find older and the device powers on, canze connects to it but that's all. Just power and BT lights come on.
this is what is inside the dongle
https://imgur.com/a/PjvxM2D

ericmini, 2023-08-09:
@fesch do you want me to post you the OBD for testing? I am also having the same issue with Konnwei kw902 that's indicate with BT 3.0
I thought the OBD is not working so I bought another one but still the same issue, so I can give you the spare one for testing if you want.
So I tried to connect by Samsung and Nokia Android phone to the OBD for checking the Zoe, both are able to connect as the OBD light keep flashing, but bottom left corner on the screen indicating fail to load the Debug. It's does keep loading for few more things and all fail. I wish the screenshot can give some insight.
[image]

fesch, 2023-08-09:
Hi @ericmini 

As far as I see on the screen shot, the application is not getting any data from the dongle. This can be either the case if the dongle follows another protocol than ELM327 original does of because of some other communication failure.

You may try to open a serial connection to the dongle using some other application or you PC ... just to see if there is data on the line.

ericmini, 2023-08-10:
> Hi @ericmini
>
> As far as I see on the screen shot, the application is not getting any data from the dongle. This can be either the case if the dongle follows another protocol than ELM327 original [...]

Hi fesch, thank you for your reply. Here is the Konnwei KW902 for your reference.
It does working fine with Torque and able to connect a Honda Jazz. So, it's either CanZE cannot connect the dongle or the dongle not suitable Renault Zoe, I don't have the clue, maybe I post you the dongle then you can do some testing? I looked in previous post from other people and looks like they are having similar issue in Konnwei KW902 with CanZE.
[image]

fesch, 2023-08-10:
Before doing that, you may try what is described on the following page:

https://canze.fisch.lu/dongle-test-script/

ericmini, 2023-08-11:
> Before doing that, you may try what is described on the following page:
> 
> https://canze.fisch.lu/dongle-test-script/

Really appreciate all your support, after following the step from described page then OBD is working fine now!

henrikx, 2023-08-11:
This issue kind of got off-topic...

@fesch Do you know anything about Bluetooth LE support?

fesch, 2023-08-17:
@ericmini : Someone else reported, that his dongle started working, once he ran the dongle test script as described here: https://canze.fisch.lu/dongle-test-script/

ZoeZE50, 2023-09-09:
Hello altogether :-)
Today I also received a KONNWEI KW902 Mini BT 5.0 OBD-II Auto Diagnose Scan Tool. I connected it to my Huawei P10 via bluetooth. Then I started CanZE selected the KONNWEI adapter in the settings, also selected the ZOE ZE40/ZE50.
But thats is, no connection to the car. ELM test was also not possible.
With CarScanner I also could connect to ELM but it was not possible to get the connection to the ECU.....
I explicitly bought the Konnwei adapter after the McLean adapter did not read all the data.
Charge info yes but firmware not. 
Now I have the Konnwei which is not able to connect to ECU.
I know, often the problem sits in front of the screen but I am lost.
Has anybody an idea before I send back the Konnwei?

fesch, 2023-09-09:
Recently two persons told me independently that their dongle started working after they ran the ACID test script from the following page:

https://canze.fisch.lu/dongle-test-script/

Maybe that is worth a try before sending it back ...

ZoeZE50, 2023-09-09:
What I have done in the meantime.
I have done this test manually with Bluetooth-Terminal every test went fine only with the "03222006" I got "NO DATA" as message.
I tried to run the ELM Test in CanZE. The initialization worked but the I got this message:
[image]

Next I analysed the ELM version of the Konnwei with "ELM Identifier". All 1.x versions are supported but not the 2.x ones
[image]

Next I tried another car (combustion engine) with "Car Scanner"
Everything worked fine. I could connet, read the sensors, get the firmwareinformations, all perfect.

Next I went back to my Zoe R135 ZE50, using "Car Scanner" just changing the car to the ZOE and what should I say. Car scanner connected to the ELM and ECU. I got the SoH Information and others. E.g. the LBC firmware from the ZOE I looked for (as assumed the old 530).
[image]

I also checked the ODBII and the ECU protocol:
[image]

**So I think the Konnwei is working well but unfortunately not with CanZE. It would be great if anybody can help me to get the Konnwei running with CanZE.**

[image]
[image]

fesch, 2023-09-09:
Is the dongle a KW902 like this one?

https://www.amazon.de/ELM-Scanner-Auto-Wifi-Diagnosescanner-Codeleser-Erkennung-12-V-Diesel-Benzinfahrzeugen/dp/B087QL3W4Z/

ZoeZE50, 2023-09-09:
I think yes, see pictures attached below.
I have tried once again with "CarScanner" there the KW902 works without any problems. I get SoH, ECU firmware e.g. for LBC & LBC2 etc.
In the CanZE App I can not change the car in the settings I only can switch within the Zoe derivates but I am not able to change the car (e.g. Megane) or the manufacturer. I deinstalled the App an installed new but still the same. Zoe seems to be preselected or let us say CanZE refers to the previous setting despite deinstallation and new installation of the App.
Might it be that I have an issue with Android version 9.0? The P10 did not get a higher Android version (end of live). Might that be a problem in combination with CanZE 1.67?
[image]
[image]
[image]

KarelSvo, 2023-09-10:
> I have done this test manually with Bluetooth-Terminal every test went fine only with the "03222006" I got "NO DATA" as message.

This is completely normal ;-) the ECU address used in the test does not exist on the Ph2 Zoe.

ZoeZE50, 2023-09-11:
@KarelSvo,
Thank you for the information. I think CanZE does not seem to be compatible with the Konnwei. Maybe because of new software / components in the Konnwei?

What can I do now to get CanZE running?

I have ordered three other OBD adapters and will try with them. I will write then how it went.

If anyone else has an idea and can help me would be very happy!

iboboc, 2023-09-11:
Hi @fesch, not sure if you understood the picture here.
Konwei KW902 BT 3.0 Classic works just fine with Canze Android app.
What is reported here is a new (another) model of Konwei, still KW902 but BT 5.0, Low Energy.

AFAIK, there is no support in Android code for BLE, as it requires different connection code (maybe I am wrong, I didnt checked out recently the Canze source code). If it is the case, then the open question is if you want to add changes for BLE connections, beside BT.

My 2 cents.

Regards,
Bobby

fesch, 2023-09-11:
No, CanZE does not yet implement BLE connections. There were some developers trying to, but I think they never finished :-(

iboboc, 2023-09-11:
I thought so, it makes sense.
The advantage will be that there will be one more device supported and the same device can be used for both iOS and Android.
But not sure it is worth it, considering the amount of effort in development. I suggest the users to just buy the right device, it will do its purpose.

ZoeZE50, 2023-09-11:
Hello all,

I received two adapters today and tested them (Android and iOS).

1. Vgate iCar Pro
The test with ELM327 identifier results --> ELM327 v2.3!
Works with CanZE 1.67
Only on the "main page" nothing is displayed, for whatever reason.
Firmware of all ECU is displayed except these two: Airbag SRS and DCM.
Interestingly, the adapter is also recognized and connected by CanZE under iOS, but all requests to the adapter end with a timeout message.
See pictures for details below.

2. Maclean MCE200
The test with ELM327 identifier results in --> ELM327 v1.4
Works with CanZE 1.67
Again nothing is displayed on the "main page".
Also here the firmware of all ECU is angezeit except: Airbag SRS and DCM.
This adapter is not recognized by CanZE under iOS.
See pictures for details below.
Tomorrow I get another and will report how it stands with the then fourth variant.
The Konnwei goes back because that does not work at all.

@iboboc & fesch: although the Vgate is a BLE 4.0 this works with CanZE under Android. That can not be the reason that the new Konnwei 5.0 does not work.

I will keep the Vgate. The only one that can ELM327 v2.3 and it works with Android and iOS, although under iOS may have to use another app such as CarScanner. 

[image]

[image]

[image]

[image]

[image]

ZoeZE50, 2023-09-11:
Sorry forgot the following pictures:
1. CanZE Android settings showing Vgate connected despite it is specified as BLE 4.0
2.+3. CanZE iOS settings showing Vgate connected 
4-6. Images of the products

[image]

[image]

[image]
[image]
[image]
[image]
[image]

henrikx, 2023-09-12:
> No, CanZE does not yet implement BLE connections. There were some developers trying to, but I think they never finished :-(

@fesch Would be great if you could update the [documentation](https://canze.fisch.lu/elm327/), so that people are aware that the BT 5.0 version is not supported. Thanks.

jorismw, 2023-09-12:
Thanks for all your updates, I am currently in a similar process and unfortunately finding out the hard way just like you. I got the same KONNWEI as you but orange. It said WiFi in the title of the ad but once I got it home it turned out to be a BT 5.0 variant...
My Vgate iCar Pro should be delivered tomorrow and then I can continue finding out how this stuff works. Thanks for your elaborate posts.

KarelSvo, 2023-09-12:
> Vgate iCar Pro

https://github.com/fesch/CanZE/issues/775

ZoeZE50, 2023-09-14:
Today I got the Vgate icar 2 BLE4.0 OBDII Interface.
The test with ELM327 identifier results --> ELM327 v2.2
Works with CanZE 1.67
Only on the "main page" nothing is displayed, for whatever reason.
Firmware of all ECU is displayed except these two: Airbag SRS and DCM.
Interestingly, the adapter is also recognized and connected by CanZE under iOS, but all requests to the adapter end with a timeout message.

All in all I keep the Vgate iCar Pro, giving me the  most information on a v2.3 lvl for the future.

I am happy when this stuff helps :-)

I am curious what my dealer says when I request the firmware update :-)

Stay tuned :-)

ZoeZE50, 2023-10-26:
Today I took my ZOE to the dealer for the upcoming service and of course the firmwareupdate that I have announced when making the appointment. Unfortunately it seems that at a big dealer only one person is familar with e mobility. This guy is on a training this week, next week he is on holiday and the week after again on a training.......
So I have to show up at the dealer at the 16th of November again.
This to keep you informed.......
Unbeleivable, w/o further comments......

ZoeZE50, 2023-11-20:
Well, today I had the appointment at my dealr to get the Firmware update of the high voltage control units. My dealer told me that he will charge me by 500-600€ to update the Firmware what I rejected. I wrote an email to Renault Germany that I see here a hidden defect that only now materializes through the reading of the SoH and is therefore to be remedied free of charge for the buyer.
Will see what happens, I keep you informed.

brietman, 2024-02-22:
Is they Bluetooth 5.0 thing still am issue? Because I also got a konwei BT5.0 version. But it just works fine.

Really glad it works. Got the zoe 50 soh-bug... 75% @30.000km..

ZoeZE50, 2024-04-04:
Hello brietman,

sorry I was a long time not online :-).
Regarding your question, the Dongle has noting to do with the soh read out via OBD. But don't worry, you can be nearly 100% sure that the 75% is wrong. 
If you start CanZE, select the tree buttons on the top left of the screen, then select in the pulldown "firmware". There you select LBC, then you get the firmware version of your LBC controller, I think it will be softverrsion 530 which is the faulty version leading to wrong (too low) soh values.
You have to turn best direct to Renault in your country and tell them that this is a well known bug leading to wrong soh. This is a hidden defect that has only now materialized by reading out the soh and checking the SW version. So the product liability applies.
It might take some discussion, as it was with my car also. Renault will send you to the local dealer that they check the SW version of the high-voltage control units. The you have to tell the dealer that the 530 version is wrong and old and all the high-voltage control units have to get an update to the newest version. Normally Renault shoul bear the cost. If they try to refuse, you have to be persistent and, if necessary, threaten them with a lawyer due to product liability. But only as a last resort, in my case they have already given in and I now have the latest SW version. Before soh equals 78% now it is at 95% at 58.000km.......

brietman, 2024-04-04:
> sorry I was a long time not online :-). Regarding your question, the Dongle has noting to do with the soh read out via OBD. But don't worry, you can be nearly 100% sure that the 75% is wrong. If you [...]

Thanks for your answer. The problem already got fixed somewhat (it now read 90.17%). The LBC was 521 (yes even older). But I wanted to be 100% sure that my reading is right.  Because both the dealer and the Renault Nederland, said: 'NO it is 100% 92% SoH.  We don't trust third party stuff and the Zoe was on the clip, so the value is 100% right, don't worry.' Then he said: 'oh wait I can see, that you have some errors on your Zoe, you can better check those with your dealer'. That I did, with other reports of strange behavior (the usual buggy stuff with the ZE50 combined with the older LBC). And they updated the LBC, LBC2 and BMC. 

They tried to install the energy info screen, but no luck the first time. Fixed the 2nd time. Also my clock isn't working anymore (it only remember the time which I set it to). 

But..... how old is your ZE50? Mine now has a SoH reading of 90.17% with 33.000km.  And is from April 2020. I expected a bit. Could it be possible that de buggy firmwares worn out the battery more? It always hang at 99% when charging for an hour or so. Or the last owner, used CCS-charging more often.

ZoeZE50, 2024-05-16:
Again sorry for the late reply. My ZOE is 10/2020 with 59.000 km. SoH 95% still :-). Charging 90% AC and 10% DC.

citroniais, 2025-05-18:
> TLDR: Recieved a BLE 5.0 version of Konnwei KW902, wanted to try the tests mentioned at [Dongle test script](http://canze.fisch.lu/index.php/dongle-test-script/) with [Bluetooth Terminal (on APKpure)](https://apkpure.com/fr/bluetooth-terminal/Qwerty.BluetoothTerminal), but wasn't working on three different phones. So i connected my KW902 to a Windows BLE 5.0 capable laptop, ran the "Test ELM" function of DDT4all, then went back to CanZE, and it worked.

Hi, I know I'm late but I wanted to share my experience with that in the case that it could be useful to someone.

As I knew BLE 5.0 version of Konnwei KW902 were not supported, I looked on ebay to buy one that mentioned BT 3.0 in the description, and no BLE 5.0 mention. It was a bit hard to find but I was happy. Despite my efforts, it turned out I recieved a BLE 5.0 version... Moreover, I contacted the manufacturer directly through Alibaba, and they say that : 

[image]

**BT 3.0 variants of their products (KW90x) are not manufactured nor sold anymore.**

So, as expected, it was not working with CanZE. I asked a refund but only had a partial one, and the seller let me keep the product. So I thought, why not trying the things said here.

I first tried to get Bluetooth Terminal to do the tests mentioned at [Dongle test script](http://canze.fisch.lu/index.php/dongle-test-script/), but it's not existing anymore on Play Store. So I looked for websites online, and I found it here : [APKpure Bluetooth Terminal](https://apkpure.com/fr/bluetooth-terminal/Qwerty.BluetoothTerminal):
- tried to install it on my Android 15 Fairphone 4, not compatible (too old app) ; 
- then tried with an old Android 7 Honor 6A, it installed and connected to the Konnwei KW902, but every time I was sending a message, it returned me the same message with a question mark (for example: >ate / ate?). Then realised it was only BT 4.2, so not compatible with BLE 5.0
- finally tried to install it on an Android 11 Redmi 9, BLE 5.0 compatible, but same behaviour as the Honor 6A (returns the same message with a question mark); so **I gave up with Bluetooth Terminal**

In the meantime I was trying to connect it to my Windows 11 Surface Go 2 (with BLE 5.0), in order to use it with [DDT4all](https://github.com/cedricp/ddt4all), but it was not showing in the menu to pair a Bluetooth device. So I installed these [Bluetooth CLI tools](https://bluetoothinstaller.com/bluetooth-command-line-tools) to play around, started a CMD as admin, and typed `btdiscovery`. It returned me all near devices, including my Konnwei KW902. 
So, next step, I forced pairing with it:
- Open Windows settings, go to the Bluetooth section (don't know if really needed but that's what I did)
- Execute `btpair -p -b[macaddress]` without space between -b and [macaddress], and replacing [macaddress] with the MAC address of the Konnwei OBD adapter found previously with `btdiscovery`
- A notification _should_ appear on Windows asking if you want to connect to "KONNWEI", click on it and it will redirect you on the Windows settings
- Accept the pairing in Windows settings
- Open DDT4all, click the Bluetooth button, and then select the adapter (in my case they were COM3 and COM4, and it was COM4 after testing both)
- Click the "Test ELM" button, a lot of tests should be done (if nothing happens after 20 sec, close DDT4all and try other COM ports)

**After this manipulation with DDT4all that ran some tests, I turned off Bluetooth on my laptop, and tried to reconnect my Android 15 Fairphone 4 to it, and then opened CanZE. And to my surprise, it works!**

So Bluetooth Terminal seems to not be working anymore (or I don't know how to use it, it may also be possible...). But ELM tests made by DDT4all also helps to make a Konnwei KW902 BLE 5.0 adapter work with CanZE.