fesch/CanZE #781: App crashes directly

TexM23, 2023-08-18:
After Google PlayStore updates (3 in a few hours/days), the app crashes directly.
Android 13 
Before the updates it ran without problems.

fesch, 2023-08-19:
This is due to permission changes in Google API. I've tried to solve it yesterday, which explains the 3 updates in a row ...
 What you might try:
 - Uninstall / Reinstall in order to reset permission
 - Give the "nearby devices" permission to CanZE

TexM23, 2023-08-19:
Both made.
After that, the app starts again without problems. But no more data comes. Bluetooth connection seems to be down.

Found an old version 1.63.50 - It works without problems.

MAlfare, 2023-08-19:
V 1.66 Andriod 8.1 Endless hourglass when trying to open settings.
Without connection to dongle/ZOE.
Where can I find less recent versions?

fesch, 2023-08-20:
@MAlfare : Here https://apkpure.com/canze/lu.fisch.canze/versions

Thanks for all your feedback. We are working on it ...

gpezzarini, 2023-08-20:
Confermo il non funzionamento.
L'ultima versione non si blocca, ma non si connette con OBD.
Grazie per il lavoro.

gpezzarini, 2023-08-22:
I test the new release 1.67, I tink it's ok, tomorrow I use the CAR to a little trip.

closed 2023-08-22 by TexM23

asez73, 2023-09-11:
Hello, since this upgrade, I can't log the debug data as I was doing before... 
This is on a Pixel 6a with Android 13 fully updated. 
If i check the log debug box, CanZE doesn't log anything and I can't get any data while my dongle is still properly connected in BLE. Unchecking this box displays almost everything excepte that the radio link seems much shorter: from the driver seat, it works... 
Finnally, an old version of CanZE works as usual for me but with a Redmi note 8T under android 11, everything is unchanged as during the last 2 years.
All apps come from Play store.

Stromer111, 2023-09-28:
Hello, I have used the tool so far and was very satisfied. Therefore, thank you very much for programming. With the latest version, there is no Bluetooth connection. Konnwei is paired with the smartphone, but remains red. Attempts to connect via CanZee, but to no avail. Also tried all the recommendations from above. Unfortunately, no success. App came from Play Store. Ask for a solution.

fesch, 2023-09-28:
With the new version of Android, Google change the permission structure, which now make a lot of problems. The app seams to work for some users, but not for all, which makes debugging very painful ...

MAlfare, 2023-09-28:
1.67 seemed to work on my Android 8.1 Motorola 4G+, but "Charge-Diagrams" did not.
So I am back on 1.63 and I will stay there forever,

AlfredR1, 2023-10-17:
I had troubles to read or write  TMP sensor with my Samsung A22 and CanZE V 1,68. After appling the hint to install V1.60 I can change the IDs from summer to winter tires and can read the new values.
Something that never worked is to store or read TMP Codes to SD. Maybe this is an Android access problem, althougt all permissions are set for CANZE. In the internal storage I found some files _TiresA.csv in, but this are not changed since 2020

AlfredR1, 2023-10-17:
Read/Write of TMP sensors does not work with V1,68. I stepped back to V1.60 - Everything is fine!