fesch/CanZE #783: CanZE to showing lowest and highest charge of modules ?

dbwarrior1975, 2023-08-29:
I found out by clip tool that there is values for these values. 
Is there any interest to add them to UI ?

Lowest is : 7bb.6103.32
Highest is : 7bb.6103.48
[image]
[image]

yoh-there, 2023-08-29:
I've labeled this as an enhancement request. Fairly easy to do, though at the moment we're still struggling with a few issues, so it won't be "immediate". Thanks.

yoh-there, 2023-08-31:
I did some checking. There are IMHO to issues that makes this less attractive than I first thought.
1) The data does not seem to be available for the LBC of the ZE50 model.
2) the actual data name is "Battery SOC of cell lowest voltage"
The latter may be nitpicking, but balancing does change the coulombs per cell. Former I feel is, with many more Ph2's now on the road, a more serious issue.