fesch/CanZE #784: Adding Balancing activity info to Voltage Heatmap view

dbwarrior1975, 2023-08-30:
Each module has configuration of 2p8s.
12 modules are connected in series.
This means you have a total of 8 x 12 = 192 cells.

LCB ecu offers balancing activity info over each cells(bleed resitors activated). 
Each hexadecimal value represents one byte, which is 8 bits. Each bit could represent the state (on/off or true/false) of each cell's bleed resistor in a module.

Let's decode your example hex values:
    01 in binary is 00000001
    80 in binary is 10000000
    BF in binary is 10111111
    ... and so forth for all values.
    
 It would be great if we some how visualize this activity on Voltage heatmap view. 
 It could be example some black border of each cell(attacment).
 What you think ?    
[image]

yoh-there, 2023-08-30:
I think this is a very good idea and will put it in the "enhancement" category. Not super complex to implement either. There is one caveat: we do not know which bits map on which cells, and that is an issue. If you have figured that out, i.e. by noticing the bleeding bits always corresponding with the highest voltages, that would be valuable info.

As UI I would propose something less "in your face" (i.e. a small black dot or triangle in a corner) but that is trivial details.

yoh-there, 2023-08-31:
I'll release this shortly in the beta channel. The indication is a subtle underline. I stress again that there is no guarantee whatsoever that the bit sequence, both on a byte by byte basis, as well as the bits within a byte is correct, and any feedback is appreciated.

I'll close the issues as it is programmed. Feel free to reopen for feedback or issues.

closed 2023-08-31 by yoh-there