fesch/CanZE #785: TPMS Read/Write not possible in V1.67

Elcamillo, 2023-09-01:
With the new update to version 1.67 it is not longer possible to write the TPMS Sensors with a Konnwei KW902 ELM327 to a Zoe Q210. There is an Error message saying that the value is empty. I did a crosscheck with CanZe Plus - there the TPMS issue is not observable. I tested this with two different KW902 Adapters and 2 different Zoe Q210. So I assume the root cause is something changed in the last Version.

1f5, 2023-09-12:
I can confirm this issue. I downgraded to 1.62 as a workaround, which worked perfectly with the same adapter and car.

yoh-there, 2023-10-02:
Can you please retest with 1.68? We had some strange issues with recent compiler optimization settings which are now turned off.

WolfgangHG, 2023-10-07:
Same problem here with 1.68.
The toast error message is "E-Empty result".
[image]

I have a ZOE 40 from 2019, but strange enough I have to switch to "ZOE R90/110" in the settings to get an ELM connection.

I don't know whether the TMPS sensors worked for me with a previous version of CanZE - started using CanZE this summer and used it only once ;-). And as my mobile phone was updated to Android 13 recently, I don't know whether older CanZE versions would work there anyway - weren't there changes to privileges recently?

Where is the debug log file placed? I don't find it here: https://canze.fisch.lu/2020/11/29/canze-file-location/

yoh-there, 2023-10-07:
I really need feedback from a few more users. While I own a Ph1 car (a Q210), it is a 21013 model and is not equipped with TPMS, which became mandatory in September 2015. I therefor cannot test this myself.

I'll get back on the log file issue

fesch, 2023-10-07:
I have a Zoe2 but disabled the TPMS as it was not working as it should ...
CanZE did not get IDs too ...
So I am of no help either :-|

DrVirvar, 2023-10-16:
Hi. I have the same problem with mye 2016 Zoe using 1.68. E-empty result. I changed to 1.62 but it did not help. (it is a build from august this year i think) If   you tell me how to do a log I can try.

DrVirvar, 2023-10-16:
i needed to go back to 1.60 to make it work. 1.62 did not.

WolfgangHG, 2023-10-17:
@DrVirvar how did you switch to the old version? I tried F-Droid, but as far as I understand it, there is only the most recent 1.68 available.

DrVirvar, 2023-10-17:
I used a tool called xapk installer from Tarrysoft on google play. It is horrible but free and it worked on my old galaxy tab. Then I found the right version av canze on apksos.com and miraculously it worked. Good luck!

WolfgangHG, 2023-10-17:
OK, I should have googled it myself ;-) - but I am quite a newbie with android and hoped that someone has a guide for this...
The "xapk installer / apksos.com" approach did not work for me. "apksos.com" versions seem to require a "Split APKs installer" which is totally outdated and does not start on recent android, and Tarrysoft "XAPK installer" could not handle the file downloaded from apksos.com.
But I found https://apkpure.com/canze/lu.fisch.canze/download/1.60, which provides a directly installable APK file. More testing with the car tomorrow ;-)

WolfgangHG, 2023-10-18:
With 1.60, I could really read the TMPS. But it is a unreliable process - sometimes, I see the IDs, and only once I got IDs and pressure.
With 1.68, i failed always.
[image]

Is it maybe a timing issue? I would like to help with debug logs, but the question for the log file location is still open ;-)

To my understanding, the process is: After having changed to winter tires, I ask my renault dealer to store the new pressure sensor IDs, and then I read them with CanZE and save the screenshot. Next spring, after changing to summer wheels, I enter the summer wheel IDs in the fields and click "ID schreiben"?
https://canze.fisch.lu/2019/01/20/tpms-ids-can-now-be-read-and-written/
What is the meaning of the "Save to/Load from A/B" buttons?

E-Gerd-21, 2023-10-19:
Same problem on my side with ZE40 from 2017. In the TPMS "ID schreiben" and "ID lesen" does not work with ver. 1.68.  It worked for many years in previous versions.

E-Gerd-21, 2023-10-19:
> To my understanding, the process is: After having changed to winter tires, I ask my renault dealer to store the new pressure sensor IDs, and then I read them with CanZE and save the screenshot. Next [...]

"Save to" will store the 4 IDs locally on your mobile phone. "A" for the summer tires, "B" for the winter tires.  There is no need to enter the IDs manually after you changed your tires. Just use the "Load from" function.
"ID Schreiben" writes the IDs, which you see on the CANZe screen into the ECU.
So, after you changed to your winter tires you need to perform first "Load from B" and afterwards "ID schreiben" in the German language setting resp. "Write ID" in the English language setting.

DrVirvar, 2023-10-19:
"What is the meaning of the "Save to/Load from A/B" buttons?"
First use the Canze tool to read your codes from your summer tires and save these to "A" (Save to A) (A is Summer :-)
Then change the tires to winter and drive to your Renault dealer and ask them to store the new pressure sensor IDs,
then use the tool again to read the new ID's and save these to B. (B is winter)
Now you have your summer tire codes in A and your winter tire codes in B and never have to go to Renault again.... (Haha)
In spring you now change to summer tires, and use the tool again. This time you press the "Load from A" button (Summer tire codes) and then the "Write" button. And then you can check if it worked by trying a "read" and see if nothing changes in your window. You should of course take a screenshot of both the A codes and the B codes in case they are lost, and you should write them down in a way where you know which actual tire has which actual code.... in case they are changed around.

WolfgangHG, 2023-10-19:
@E-Gerd-21 @DrVirvar Thanks for the clarifications! But I fear I will have to wait for a CanZE fix so that reading the sensors works with 1.68. Even 1.60 was not reliable...

It would be great if the IDs could also be exported to/imported from a file so that I have a fallback if the app settings get lost - for example when unstalling the current version to switch back to 1.60 in order to test something ;-).

Demolin-L, 2023-10-21:
Still not working in 1.68 on wifes Zoe R90 2018.
Was wondering what do to, then found a new app on google play that seems to be pretty much a copy of this one called CanZE Plus.
It works fine and exactly the same way in that one, just have to store your A/B presets again.

I assumed this app had been abandoned and these new guys had taken over with the new app, glad to see that is not the case, hope this one works again when i switch back to summer tires next year!

fesch, 2023-10-21:
No, CanZE plus is just a fork of a earlier version of CanZE ... but, hey, if it works for you, than use that one ;-)

Actually we are working on a Bluetooth permission problem, which is kind of a problem, because neither settings / permissions we use, none seam to work for all versions / users.

This is certainly not related to the TPMS, but just an explanation about the actual situation ...

oddeirik, 2023-10-21:
Can confirm that the latest version on Play Store does not work when trying to read or write TPMS in an R240 on a Samsung Galaxy S10e running whichever Android is the latest one available from Samsung. 

Tried it multiple times today, reinstalling the app from Play Store, toggling permissions for the app, nothing, just the "Empty result" error. I ended up using the 1.60 version from apkpure linked further up and it worked fine. Well, as fine as it can given that I am using a possibly-Konnwei-kw902-ELM327 adapter... 😅

Elcamillo, 2023-10-22:
Today I tested again with CanZe 1.68 - same Problem as with 1.67. Reading/Writing TPMS values not possible. It works with CanZe Plus 1.63.50 - unfortunately I got error Messages here when writing TPMS IDs, but they are written anyway.

zingmane, 2023-10-28:
Same problem here with v1.68 and ZOE Q210

DrVirvar, 2023-11-02:
When actually wanting to use the 1.60 it had of course updated overnight. I then downloaded the Canze plus from google play which worked fine. In fact it seemed much more stable (Bluetooth) than any of the versions of Canze that I have used before.
So plus it is.

redflo, 2023-11-11:
Same issue here. Maybe it is a App permissions problem? The App does not have or request the permission to read/write to internal Android storage, and cannot read the stored TPMS values. If i try to switch on debug log, the TPMS App page is greyed out completely. Using OnePlus Nord 2 on Android 13

Dooleweerdt, 2023-11-12:
Same issue here. Using a home made CanSee dongle, Chromebook and latest CanZE app version 1.68.
Workaround is to use the CanZE Plus app, as mentioned by others. Actually it seems more stable :-)

dominiquedemunck, 2023-11-25:
Renault ZOE from 2017 here (R90).
I tried reading out my stored tire sensor ids with version 1.68 and my Fairphone 3 (Android 13) but immediately got "Can't access file/storage/emulated/0/CanZE/TiresB.cvs". Also had TPMS empty errors but other things just work.
I also noticed that when turning the logging on, I cannot see any of the outputs anymore (battery state, ...), they all turn into N/A, 
When I go looking into the internal storage (Android/data/lu.fisch.canze), I only see older files from before my Android upgrade.
Tried removing those folders and all app data and reinstalling CanZE but all these issues remain.
So given these facts, it might still be a permission problem? See also this older post:
https://github.com/fesch/CanZE/issues/626

Then I used CanZE Plus on an older Samsung Android 8.1 version and there everything just works :).
That app isn't available for Android 13 I guess since it doesn't pop up in the Play store.

mekholm, 2023-12-17:
I can confirm it's bigger than just TPMS.
I couldn't get anything to work with 1.68. It would connect to Bluetooth, only to loose the connection as soon as I tried to read or write anything.
Couldn't read or write the stored A/B TPMS IDs, or write the logs.

Inspired by the discussions here I tried Canze Plus to try the older version. Everything worked directly.
But I'd much rather use CanZe, since the plus won't run on my phone.
Happy to help if there's something I can test or clarify!

martindlarsson, 2024-05-29:
Any update in this issue? I tried CanZE Plus and I was able to install it on my Pixel 7a with Android 14. It worked after a few tries.

WolfgangHG, 2024-08-11:
I tried CanZE Plus, but it did not work at all for me. With CanZE I could read the battery SOC, while CanZE Plus could not access any information (Android 13).

GallusMax, 2024-09-16:
Same for me, version 1.68. I am sure to have read the tpms serial numbers before. Any progress?