fesch/CanZE #786: No graphs in the last release (1.67)

Amazit, 2023-09-07:
Hi!
I installed and checked last version of CanZE.
Now connecting to dongle work properly but no graphs are showing in app.

CanZE 1.67, all permition accepted.
Samsung s10e, Android 12

Additional info on screenshots
[image]
[image]
[image]

yoh-there, 2023-09-07:
I can confirm this, Samsung A13. The frustrating thing is that it works fine when compiling locally and installing. The problem occurs only when going through Google Play. Reaching out to @fesch  as I am baffled!

Eidyn17, 2023-09-17:
Same here. Samsung S23 plus.

Telemuddi, 2023-09-27:
Xiaomi 12 same problem.

fesch, 2023-09-28:
It is strange that some values are present, while the graphs do not fill. This means that the dongle sends some data but not all ... !?

yoh-there, 2023-09-28:
I don't think this is a dongle problem. Same on my esp32 CanSee dongle. The graphs do not initialize ie stay white in dark mode. And frustratingly, it works fine when compiled locally and installed directly without Google Play. I can guess I can help out if you want to release in the beta channel?

fesch, 2023-09-28:
let's give it a try!

yoh-there, 2023-09-28:
Sure, let me know please once you've pushed something on the beta channel, or if I should pull - compile - install on my phone

fesch, 2023-09-28:
Just pull & compile. I did not change any code the last few days ...

yoh-there, 2023-09-30:
We might have found the issue. I have released 167beta2 on the open test channel and will give it a spin the next few days. If successful we'll push it to production of course.