fesch/CanZE #788: CanZE support for USB ELM327

camilcll, 2023-09-24:
Hello,

Would it be possible to use canZE with a USB ELM327 plugged in a computer? I might start looking at the code to see if I can contribute on this point.

fesch, 2023-09-28:
Actually CanZE open the dongle via BT classic, thus serial.
Writing a new driver for serial over USB should be easy and quite straight forward....

yoh-there, 2023-09-28:
I know I am kicking in an open door, but you can only do this on a computer if it's emulating an Android device, so basically using the Android Studio environment, or maybe a Chromebook.

I couldn't get data over BT working on CanZE on my Chromebook BTW.

fesch, 2023-09-28:
Windows 11 has Emulation for Android apps integrated :-)
But connectivity was very poorly implemented last time I tried it ...