fesch/CanZE #789: New supported language - HU

Adamyno, 2023-09-29:
Hello.
I made a Hungarian translation for the app. 
I don't know how to technically get it to you, so here's a link to the text file.

[https://www.toptal.com/developers/paste-gd/wy13rHFH](https://www.toptal.com/developers/paste-gd/wy13rHFH)

I thought a lot about how to translate the technical terms. I mostly left the error messages in the original language so that it would be easier to find a solution.

I would try it in practice and it is possible that based on further feedback, I would fine-tune the language file.

yoh-there, 2023-09-30:
Thank you! I will add the file to the repo and release it in Open Testing soon

yoh-there, 2023-09-30:
I added the file to the repo (development branch) with attribution to you of course, after removing the entries with the untranslatable attribute.

Adamyno, 2023-10-05:
I tested it and fixed some bugs. 
I downloaded the source code and finally compiled it myself so I could test it again.

There are some text that appears to be hard coded.

For example, under the main fragment:
"Now ZE50 enabled!"

Or under battery activity:
"Delta with reality (km)", "Range (km)"

Auxiliary battery:
"Voltage under load", "Current under load", "Voltage", "Vehicle state" (on the diagram)

Tires:
"Save to A", "Save to B", "Load from A", "Load from B", "Swap front rear"

The value of "button_speedcontrol" is not visible, the label of the button always remains "AVG SPEED".
[values-hu.zip](https://github.com/fesch/CanZE/files/12823090/values-hu.zip)

yoh-there, 2023-10-06:
Thanks. I'll have a peek at the issues mentioned this weekend. Reopening.

yoh-there, 2023-10-08:
I included the new file and fixed the issues you mentioned in the development branch, thank you!

Let me know if you require it to be compiled to the Play store in open testing for you to test please.