fesch/CanZE #791: CanSee PCB design

krekr, 2023-10-05:
Sorry to add this as an 'issue', but I can't really find any other place to post this. I'm having a hard time to find either a WiFi version of the ELM, or one of the other iOS-working dongles. Most sellers just refund me, tell me they're out of stock, or send the wrong version. I was thinking of making the CanSee dongle, but it might be worthwhile designing a PCB for it so its easier to use.

It's not worth the effort just for myself, but maybe a lot of other people are interested as well? I might put it up on Tindie when it's done. Anyone interested?

yoh-there, 2023-10-05:
First of all, it's probably better to post this in the CanSee issue tracker https://gitlab.com/jeroenmeijer/cansee/-/issues

I have a few copies of a PCB laying around donated by a user. I am happy to send you one.

krekr, 2023-10-05:
Right, I was a bit confused by the gitlab interface... If there already is a PCB design, I'd love to build that; no sense in designing something that already exists :)
Thanks for the offer of sending one! Is there a way to dm you an address?

yoh-there, 2023-10-05:
Are you on speakev? I am "yoh-there"

krekr, 2023-10-05:
I just signed up but can't dm yet. Anyway, my address is already public on my website so... 
I just found the docs in the code repo... I can paypal you the postage if you want?

G. van Egdom
Holtenstraat 15
2541AW Den Haag
Netherlands

yoh-there, 2023-10-05:
No worries, I am in NL. I will delete your post. BTW, I am on the nl Renault forum too, zoe2goe.

closed 2023-10-06 by krekr