fesch/CanZE #792: Bluetooth support for Konnwei in iOS

krekr, 2023-10-06:
I just got confirmation that the WiFi version isn't produced anymore and will not come back. This is after receiving and returning 3 WiFi-versions that turned out to be BLE or Bluetooth. 

This might be something to add to the "Hardware" documentation at https://canze.fisch.lu/hardware/ to prevent others falling into this rabbithole. I will be making a Cansee, but others might not be able to. Is BLE/Bluetooth support planned for the iOS version? I'd be happy to help testing and debugging.

[image]

yoh-there, 2023-10-06:
@ik2wxx Should we stop mentioning the KONNWEI WiFi for iOS, and maybe advise that BLE dongle you have tested?

ik2wxx, 2023-10-10:
OBD dongle I've tested:

VGate (BLE), 2 different models.
LeLink (BLE)
CanSee (WIFI)
Konnwei (WIFI)

[image]
[image]

closed 2023-10-30 by ik2wxx