fesch/CanZE #793: SOH on battery page

stevegorst, 2023-10-08:
The SOH is not displayed on the battery page and the debug message says NAN.  Using the iPhone version of the app on a R110 42kwh battery.
[image]
[image]

ik2wxx, 2023-10-09:
Hello, what OBD dongle are you using?
Looks like some other data can't be received correctly....

stevegorst, 2023-10-09:
Hi Roberto

Its this one

Vgate iCar Pro BLE 4.0 OBD2 Diagnostic Tool Fault Code Reader OBDII
Compatible Car Adapter Check Engine Light for iOS, Android and Windows
(New)

One of the guys on the Zoe forum recommended it so I hope it's the right one.

Steve

ik2wxx, 2023-10-09:
Hello
Yes it's the model I have (two, one orange and one dark transparent). Got them in 2021 and 2022. They are running with CanZE (on a ZE50 Zoe) with no problems.
I can't be sure that current model is equal to 2021 and 2022 model.
BTW, was the car fully on? That is required for some systems to reply correctly.
Also, sometimes disconnecting the dongle and reconnecting, will solve connection problems with car.
Please try and report.

stevegorst, 2023-10-09:
The car is a 2018 model.  Yep it’s fully on and I’ve tried plugging in and
disconnecting.   I’ve also tried using Zoe 40/50 as the car but
that doesn't work at all.  I do get a SOH health on the charging screen but
not on the battery screen.  It’s lower than I expected at 86% so I’m
curious about it being correct.

Steve

asdfasdfasdfasdfasasdf, 2023-10-09:
I just got my OBD dongle and just installed CanZE.

Car: Zoe R240 from 2015.
OBD dongle: Vgate iCar Pro BLE 4.0
CanZE: 1.0.1(19)
Phone: iPhone SE 2020
IOS: 16.5.1

I can't see the SOH value either.
[image]

ik2wxx, 2023-10-10:
> The car is a 2018 model. Yep it’s fully on and I’ve tried plugging in and disconnecting. I’ve also tried using Zoe 40/50 as the car but that doesn't work at all. I do get a SOH health on the [...]

I don't have a solution right now, can't understand why it's not working.
Maybe it's a different OBD firmware...
Maybe can you try a different dongle?

ik2wxx, 2023-10-10:
> I just got my OBD dongle and just installed CanZE.
> 
> Car: Zoe R240 from 2015. OBD dongle: Vgate iCar Pro BLE 4.0 CanZE: 1.0.1(19) Phone: iPhone SE 2020 IOS: 16.5.1
> 
> I can't see the SOH value either.

Same problem with same car and same OBD (and same app :-) )

> 
> As an unimportant side note: It would be nice if you added a small amount of margin to the "Battery" page. All other pages have a small margin.

Thank you for suggestion, will do.

> As another unimportant side note: The text on the "Battery" page that says "More detailed information about these data can be found in the Q & A section of the CanZE homepage" should be removed or revised. I have looked there but I can't find any such information.

Ops, you're right. 
@yoh-there where have those informations gone?

closed 2023-10-10 by ik2wxx