fesch/CanZE #794: BT connection issue

folkdanskung, 2023-10-25:
Greetings,
I bought a dongle of a danish supplier earlier this year specifically for use with CanZE. I haven't been able to connect to the dongle via BT though. The dongle fires up fine and I can pair with it, but CanZe doesn't seem to recognize it. I tried with a generic ODBII app (Car Scanner on Play Store) and it connected to the dongle immediately. It wasn't able to connect to the ECU though (shows connection status separately for both).

So I tested the dongle in my other car, a 2013 Skoda Superb MKII, and Car Scanner connected immediately both with dongle and ECU. Everything seems to be working there, could read fault codes and so on.

So I disassembled the dongle and it was the new design with the 120 Ohm resistor, at 119-something Ohms.

Any ideas? Or should I just buy the one you linked to in this post?
https://canze.fisch.lu/2017/09/14/dongle-trouble-5/

Best regards
Marc

Version: 1.68 (release-master)
Build: 2023-09-30 12:52:47

Vehicle: 2016 Zoe R240

Konnwei KW902, BT 3.0 (looks exactly like the green one you have pictured under Hardware)

fesch, 2023-10-26:
Hi

Did you try this?
https://canze.fisch.lu/dongle-test-script/

folkdanskung, 2023-10-26:
> Hi
> 
> Did you try this? https://canze.fisch.lu/dongle-test-script/

No, I haven't, will do! Thanks for the tip!

folkdanskung, 2023-11-08:
> Hi
> 
> Did you try this? https://canze.fisch.lu/dongle-test-script/

Sorry, been busy, but managed to get time for this today. Got it to work, the dongle passed the tests and was able to connect to the Zoe with CanZE. During these last two weeks I replaced my old phone (for other reasons) which wasn't running the stock os, so may have been that.

Thank you for your quick response earlier and sorry for responding late. I unfortunately don't have anything else to add with regards to how this was solved.

Best regards
Marc

closed 2023-11-08 by folkdanskung