fesch/CanZE #795: Not able to read the serial number of the TPMS from last update

mickbl, 2023-11-09:
I'm using the app for several years, to modify the TPMS sensors on my Zoe car, each winter/summer without any problem.

I use a OBD KONNWEI KW902 (works fine until now!)
Car ZOE Q210
App version 1.68

When i accesss to the tires menu, i did "read" as previously, but all the TPMS sensors stay at "000000" (and I have correctly the pressure value in the same time).
And even if i try to write the correct TPMS value, i have an error message.
When I read:
"Could not read TPMS sensors: -E-Empty result" and all the sensors are at 00000 (but with value displayed and before swith my tires, i've checked the pressure of my Summer tires a few weeks ago with any problem)
And when I try to write the new sensors and try to replace the 00000 by the correct value:
"Could not write TPMS Valve 0:-E-Empty result"

Do you have any idea to solve it?

Christian-PK, 2023-11-10:
I am having the same issue. I am though trying on a new pixel 8 pro. Previously I used an one plus 8 pro.

piratk, 2023-11-11:
Samt issue here, it also looks like I lost the old tpms data, a bit of a bummer!

piratk, 2023-11-11:
This is a duplicate of issue #785