fesch/CanZE #796: CanZE with dongle OBDLink-CX

McSoda, 2023-11-25:
Hi All,

Not a bug report nor a request, just looking for some advices.

CanZE v1.68 (release-master)
car = ZE50 R110 / year 2020
dongle : OBDLink CX (BLE dongle)

I can't get CanZE working with the dongle. Trying any function returns a message stating dongle is disconnected.

I ran the commands with a bluetooth terminal and got:
```
> atz
ELM327 v1.4b
> ate0
OK

> ats0
OK
> atsp6
OK
> atat1
OK
> atcaf0
OK
> atsh7e4
OK
> atfcsh7e4
OK
> 03222006
NO DATA
> atcra699
OK
> atma
(nothing)
> atma
STOPPED
```
Complete log:
[serial_20231125_122423.txt](https://github.com/fesch/CanZE/files/13464087/serial_20231125_122423.txt)

Thanks

KarelSvo, 2023-11-25:
This can address is only available for Zoe phase 1.

moldav20, 2025-03-12:
Hi everyone!  
I also bought an OBDLink CX, and it works great with ABRP, but CanZE won’t connect to it.  
Is there a solution, or should I just sell it and go back to my old OBD scanner?