fesch/CanZE #797: Where do the assets come from?

ljames28, 2023-12-10:
I'm looking to find more information about the free frames exchanged on the LBC bus. I wondered where all the assets files originally came from, for example https://github.com/fesch/CanZE/blob/master/app/src/main/assets/ZOE_Ph2/LBC_Fields.csv

Thank you for any information you can provide

KarelSvo, 2023-12-10:
What you linked are ISO TP parameters.
ISO TP and Free Frames parameters are converted from the ddt2000 Ecu database.

ljames28, 2023-12-10:
Great, thank you I will look into that

closed 2023-12-10 by ljames28

yoh-there, 2023-12-10:
Above answer is only very partially correct. Here's the back story.

In the prehistoric days of CanZE, two authors did all their own research starting from nothing, using home build CANbus scanners based on Arduino Due hardware. Of course this could only cover sniffing and therefor just "free frames". This was painstaking work flipping switches, opening doors, driving around, etc and see what bits and bytes changes.

After about a year "someone friendly" provided us with the complete documentation of all free frames and which modules provided which frame. Let's say I have no idea who send it and what the source was.

The ISO-TP formatted commands we only very carefully tiptoed in later as it involved sending data on the main CANbus, which we were pretty careful about as a lot of screens were intended to be used while driving.

To my knowledge, DDT2000 was the previous generation diagnostic tool for the dealerships. This was later converted to their current tool called CLIP, which is far more locked down than DDT2000 ever was and with more "phone home" functionality.

Still, it seems some people are able to lift the definition data from new CLIP versions: new versions usually popped up on shady Russian websites first.

Those pretty large XML files are directly used by the really great open source project DTT4ALL, which many enthusiasts use for static and diagnostics use (not CanZE's focus). We have taken those files and condensed them through automation to the form found in the source code.

Hope this helps.

Closing the issue as answered.

ljames28, 2023-12-10:
Ah that's interesting. Before seeing the assets, I originally assumed all the data had been hard won by scanning the CANbus while driving to see what changes. I wonder did the "friendly" person provide free frame information for Zoe_PH2? 

I'm working on a battery storage project using Zoe PH2 52Kwh batteries. It is mostly functional and all useful information can be queried with ISO-TP. However, something is stopping the battery from balancing, so I'm trying to understand the free frames on the HEV-CANbus to see if they are having an effect. The DDT2000 CANbus frame data sadly only explained very few of the frames.

There is some information on the project here https://openinverter.org/forum/viewtopic.php?t=3807