fesch/CanZE #798: LELink not working on iOS

StuartZE40, 2024-03-24:
Hi, Have recently bought an LELink OBDII to use on my Zoe ZE40 and downloaded Can ZE app from App Store on my iPad (10th Gen). App won’t work if set to Zoe ZE40/50. Constantly times out and no data displayed. Will stay connected when set to ZoeR240 but most data missing or corrupted. Eg battery charge constantly reads 12% when actual charge is 60%. 
Settings page attached. Any suggestions?
Cheers
[image]
[image]
[image]

ik2wxx, 2024-03-25:
Hello
Please experiment with other Zoe models, use "Test dongle" button (upper right corner in the Settings screen),  it should tell you battery charging level and car's VIN.
If any model will work, OBD is not compatible.
Also, disconnect dongle from OBD port so you reset connection.
Hope this can help...

StuartZE40, 2024-03-25:
Hi Roberto,Thanks for the advice. Have tried the dongle test on all models, resetting the dongle each time. The test shows fail on all models except R240 where it shows the same 12% SOC and no  VIN. From your earlier mail would this mean there is a problem with the dongle?Cheers,StuartSent from my iPad

ik2wxx, 2024-03-28:
Hi, sorry for the delay but I wanted to check with my car and my dongles.
I have LELINK and VGATE and they worked 100% similar.
I have a ZE50 car but I don't think that's the problem:
I've tried with LELINK OBD first, and the test failed (tried several times and log OBD messages to file so I can check on my PC later).
Tried with VGATE OBD and it worked at first attempt.
Difference in app between LELINK and VGATE it's just the BLE address so the app is sending same commands with both OBDs.
Reading logs I discovered that LELINK does connect and speaks to the car but some commands do not get correct reply from car, as VGATE do.
Maybe something changed in my car firmware ?

LELINK

```
> 'atfcsh18dadaf1'
<< 'OK
>' (5)
> '021003'
<< 'CAN ERROR
>' (12)
< 'CAN ERROR' (9)
ERROR

```

VGATE
```
> 'atfcsh18dadaf1'
<< 'OK
>' (5)
> '021003'
<< '065003003201F4AA
' (17)
< '5003003201F4AA' (14)
01010000000000110000000000110010000000011111010010101010 (56)
<< '
>' (2)

```

That is unexpected to me as LELINK OBD has always worked without problems.
As a quick fix....I suggest you to get a VGATE OBD...
Sorry for that.