fesch/CanZE #802: ZE50 iOS

pvpowered, 2024-11-10:
Hi, I found a couple of videos online that showed the use of CanZE with iOS and ZOE ZE50. However, after I installed the  CanZE version from the App Store, no R135 could be selected. Testflight from CanZE4iOS seems to be expired? How can I monitor the newer models with iOS at the moment?

ik2wxx, 2024-11-11:
Hello, App Store version is ok for all Zoe models (it's the same app version that was on Testflight).
Please select correct OBD type and Zoe model and test connection.

closed 2024-11-11 by ik2wxx

pvpowered, 2024-11-11:
There is no option in the dropdown menu (Settings > Car) to select R135. Found Zoe Q210, R240, Q90, R90 - no R135 or R110. Could you please explain what I do wrong?

ik2wxx, 2024-11-11:
You should select "ZOE ZE40/ZE50"

BTW Are you sure you downloaded the correct CanZE App?
https://apps.apple.com/app/canze/id1547997741

pvpowered, 2024-11-11:
Oh my god. I did not know there were two apps in the Apple App Store with almost the exact same icon and the exact same name. Thanks a lot! Would explain why so many people complain about the stability and use of CanZE in online forums.

ik2wxx, 2024-11-11:
Only one app is official.
Please let us know if your problem is now solved ;)

pvpowered, 2024-11-11:
I don‘t know how I got the wrong app. Maybe it was linked somewhere. Yours has better reviews :D and I will add one as well as share the wisdom among my friends. This inofficial one really drags your reputation down. I‘m sorry for that. Yes, the issue has been solved - at least I found your app where I can select ZE40/50 :) need to test it later with my wife‘s car.