fesch/CanZE #807: Leakage currents measured in ze50

dijomi, 2025-09-14:
Hello, I have a question about the readings with a Canze for leakage currents

[image]

[image]

. Can someone explain to me what these readings mean. Because the real problem for a few days is that I have a large leak from the 12v battery after the car goes completely to sleep after about 2 minutes.
Measured with a multimeter about 4.5 A. A 12v battery is discharged in 3-4 hours.
The car is a Renault Zoe Ze50 2020.
Another thing I noticed is that my app My Renault is also not working, its subscription is paid.

yoh-there, 2025-09-14:
Leakage current is current in different frequency domains that leaks away to protective earth  while the car is charging. It is almost certainly unrelated to the issue you describe. Current from the 12v battery should drop significantly after you lock the car and wait for roughly 6-7 minutes; it falls off in stages.

Assuming you waited with that measurement, seems like one or more computers are not shutting down. My advice would be to check all your locks, especially the from nose lock. I had this problem once and the cause was the car locking the charging plug, but the switch into lock did not indicate a lock status. The car kept trying which bled the 12 battery. It was noticeable as I could hear the lock motor retrying over and over again.

closed 2025-09-14 by yoh-there

dijomi, 2025-09-15:
Thanks for the answer, but I waited more than 20-30 minutes with the multimeter connected to measure current and the leak is constant. The car is locked and I don't hear any sounds from anywhere. Rather, there is probably a connection with the TCU module because my Renault application does not work. It is dated 21.08.2025. and does not update.